maduce/fosscad-repo #2: This says it includes the Kisslicer ini settings but it does not...

obja, 2016-04-22:
I see the ones for Nylon - would you please commit the Kisslicer settings for ABS as well?

maduce, 2016-04-25:
the NS is for nylon and the GS was tested with ABS. cb didnt give us kisslicer files for the GS but I can ask him the next time I talk to him.

obja, 2016-04-25:
Thank you that would be great if you would. I have had pretty o.k. success with it just by using the settings mentioned. The other very ODD thing about the files is the front take down pin/pivot pin holes are not in the file and the posts where one would drill seem small. I am guessing the expectation is to drill them out but not real sure yet. Thanks though for all of your work.

maduce, 2016-04-26:
Yes, if the holes were not in the cad cb drilled them after printing. I didn't design this specific lower so I am not entirely sure why he did this. 3Dprinters can be weird about holes depending on the printer so I suspect cb just added the holes after the fact to keep the print more solid around the take down and pivot pin area.  I will definitely ask the next time I talk to cb.

closed 2017-11-19 by maduce