maduce/fosscad-repo #5: Pla or abs for bump stock

coyotebush2384, 2018-02-24:
Wondering if pla would hold up or does this need to be printed in abs

BradGroux, 2018-02-26:
PLA works fine, just remember that it can shrink somewhat so you may need to adjust for that.

closed 2018-02-26 by coyotebush2384

jimilinuxguy, 2018-04-28:
What orientation are you guys printing?

maduce, 2018-05-16:
We printed it in PLA except the stock pad part which was printed in ninjaflex. We printed it in several pieces and the orientation was in such a way that it would fit in the printer envelope. The rear stock piece was printed so the layers go vertical when you're using it...so it would've been laying butt end down on the bed. 

[image]