maduce/fosscad-repo #14: Script to download the latest official FOSSCAD repo

blackfuel, 2018-08-02:
Simple Linux script to download the latest official FOSSCAD repo to a local file for backup and mirror support, or for distribution purposes.

Downloaded files are named by the current FOSSCAD release version and the git commit hash, while retaining the git commit timestamp on the file itself.   I run it once a month using cron, just in case.  :)

**fosscad-download.sh**
```
#!/bin/sh
# fosscad-download.sh
# Download the latest official FOSSCAD repository to a local file, for distribution purposes. Compressed archive is prepared by the Github site and downloaded by this script.

set -e
set -x

VERSION=$(wget -q -O - https://raw.githubusercontent.com/maduce/fosscad-repo/master/README.md | grep -A 3 "# Version" | grep -E '\*\s+[0-9]+\.[0-9]+\s+.*' | sed -r 's/\*\s+([0-9]+\.[0-9]+)\s+.*$/\1/')
[ -z "$VERSION" ] && echo "cannot retrieve the release version number" && exit 1

SOURCE_VERSION=`git ls-remote https://github.com/maduce/fosscad-repo | grep HEAD | cut -f1`
[ -z "$SOURCE_VERSION" ] && echo "cannot retrieve the git commit hash" && exit 1

SOURCE_TIMESTAMP_X=$(wget -q -O - https://api.github.com/repos/maduce/fosscad-repo/git/commits/${SOURCE_VERSION} | grep -A 4 '"author": {' | grep '"date":' | cut -d'"' -f4)
SOURCE_TIMESTAMP=$(date -u +@%s -d "$SOURCE_TIMESTAMP_X" 2>/dev/null || date -u +@%s -D %Y-%m-%dT%H:%M:%S%Z -d "$SOURCE_TIMESTAMP_X")

FILENAME="FOSSCAD_MEGA_PACK_v${VERSION}+git-${SOURCE_VERSION}"

[ ! -f ${FILENAME}.zip ] && wget -O ${FILENAME}.zip https://github.com/maduce/fosscad-repo/archive/master.zip && touch -d $SOURCE_TIMESTAMP ${FILENAME}.zip

### [ ! -f ${FILENAME}.tar.gz ] && wget -O ${FILENAME}.tar.gz https://github.com/maduce/fosscad-repo/archive/master.tar.gz && touch -d $SOURCE_TIMESTAMP ${FILENAME}.tar.gz
```

maduce, 2018-08-09:
This is a lovely script but if you have bash you can get git. Doesn't git defeat the purpose of this script, i.e., git pull?

blackfuel, 2018-08-13:
Yes, git is the best way to stay in sync with the fosscad repo, for desktop computer users.  

However, this script can be run on devices with limited system resources that are unable to run the git client or perform data compression on a large archive.  This script is run automatically on my home router/NAS.  So, if a guy doesn't have the git client, this script would be the next best alternative.

maduce, 2019-03-21:
Good point and fabulous work.