maduce/fosscad-repo #18: Too long for most cheap printers

nickshl, 2018-10-24:
Most cheap 3D printers has area of 20x20 cm (7.87x7.87 inch) It would be nice to have version with short barrel that can fit into that area.

machond, 2018-11-10:
Okay so what minimal print area needed for now?

dorkmo, 2018-11-11:
the diagnal is about 28cm if that helps

goodwolf, 2019-02-12:
I'll have to correct you here, most cheap printers have a print area of 8" x 8" x 7". The Wanhao Duplicator i3 and i3 Plus are one of the most popular machines which are recommended in the 3D printing communities I'm rather involved with or am the owner of like @reprap on Telegram.

If the models need to be reduced then there should be README on how to slice the models for people with smaller print areas.

maduce, 2019-03-21:
if you are printing in ABS you can always print it in pieces and use abs juice to glue them together.

AT4-Backblast, 2019-06-23:
I am in agreement with @goodwolf on this matter.

Every year printers get cheaper with larger print volumes.  My current "Large" printer is 8"X8"X9" and I paid $200 for it..new.

If anything needs to be changed it's people adding STL's with parts that have a "Built-in" raft.  This is annoying and pointless.

maduce, 2019-07-11:
Some developers put printing notes in their READMEs and some do not. The models do not need to be reduced for everyone. It falls on the maker to figure out if the model can be printed on their printer. The developers have no idea what the constraints of the makers is in every situation. There is an expectation that people know how to use their printers, slicers and if necessary CAD software.

Kervi2K, 2020-02-22:
if OP is concerned about AR lowers he could just print the bolt one, it can be printed in any (I think).