maduce/fosscad-repo #20: Beretta 92fs stl is broken.

giantpune, 2019-04-09:
The STL file for the Beretta 92fs frame displays incorrectly with missing faces in every STL viewer I tried including github's viewer, tinkercad, and simplify3D.

maduce, 2019-05-20:
Sorry to here about your issue. You should be able to generate new STL files from the IGS file ( 	Beretta92FS.igs). If you install FreeCAD (or just about any other CAD program), you can open the IGES file and export it to an new STL. STEP and IGES files are open ISO formats that most CAD programs will open (which is why we try to always include them).

Hope that helps.

giantpune, 2019-05-20:
I tried that.  Freecad still produces a malformed STL file.

maduce, 2019-05-21:
Have you tried any other CAD software or attempted any STL repair tools. [Netfabb](https://service.netfabb.com/login.php) is one such tool. Meshmixer is another. [Here](https://repairkb.com/file/stl/) is a article describing such solutions. Please let us know if that helps.

giantpune, 2019-05-21:
Yes.  I tried 4 or 5 different programs and online services for STL repair.  They seem to all recognize that the STL is broken but none of them so far have been able to repair it correctly.  The ones that "fixed" it spit out a STL with big chunks missing out of it.

I'm not a super advanced user of these different CAD software.  So maybe somebody with more experience knows how to fix it.  I'm basically opening the different files from this repo and then finding the "repair wizard" in the CAD software and letting it run.

jboyscode, 2019-05-24:
what CAD softwares are you using / have you tried?

giantpune, 2019-05-24:
I tried a few of the online STL repair websites on Google search for "STL repair online", like https://www.formware.co/onlinestlrepair

Also tried the suggestions [here](https://all3dp.com/1/7-free-stl-editors-edit-repair-stl-files/).  Freecad, blender, sketchup with plugins, meshmixer, and meshlib.  And on my own, I found simplify3D's repair and tried importing it into tinkercad.

jboyscode, 2019-05-24:
have had some success with issues like this using meshlab, but I will take a look at the files when I get to an outlet, battery is low -_- I'll try to get back to you today at some point though.

giantpune, 2019-05-24:
Maybe its something you can fix without much trouble.  My mesh repair skills are limited to finding a wizard and clicking a button.