maduce/fosscad-repo #21: G17 DIY Rail

akneipp, 2019-06-03:
Left_Rail_Section_Drawing_v3.pdf, the left rail shows a 1.16mm dimension for the slide mating surface, the matching dimension on Right_Rail_Section_Drawing_v2.pdf only shows .16mm. I believe these two dims should match.

[image]

maduce, 2019-07-11:
Yeah there shouldn’t be two versions of the rear rail files and the left and right should both be at V3. Working on getting this fixed in the parts folder.

maduce, 2019-07-12:
Right_Rail_Section_Drawing_v2.pdf has been corrected and Left_Rail_Section_Drawing_v2.pdf  was removed because its an old spec sheet. 

Thanks akneipp!

closed 2019-07-12 by maduce