maduce/fosscad-repo #30: songbird frame stl incomplete

aandrich3, 2019-10-24:
the frame stl for the songbird .22lr is missing the holes required for the hammer pin to stick out of the frame on both sides.

aandrich3, 2019-10-27:
i have discovered that you can actually affix the rubberbands around the back of the hammer and the stretch the other end around th efront of the frame underneath the muzzle.

grendal87, 2019-10-27:
If you measure what needs to be where I can modify for ya.

aandrich3, 2019-10-30:
> If you measure what needs to be where I can modify for ya.

im not quite sure how to make the measurement, its a curved circular channel that spans in the x axis from the forward most position of the hammer to the back most position of the hammer.  In the y dimension, it follows the path of the top hole on the hammer.  the diameter of the hole would be slightly larger than the diameter of the top hole of the hammer.  Idk if that is enough description or not.  This channel is not integral to the operation of the pistol however it is a better execution of the cocking mechanism so its not of huge importance that the hole is there.