maduce/fosscad-repo #34: Better mirror all over while we can

ilikenwf, 2020-01-27:
https://abc7ny.com/politics/coalition-of-states-sue-over-rules-governing-3d-printed-guns/5875106/

grendal87, 2020-01-27:
Mhmm. If theres any updates do not hesitate to give me a shout. I'll create a repository of whatever files y'all got. Ban or no ban.

ilikenwf, 2020-01-27:
Good to know. In my case I think I only have this repo, the original
defdist repo, and the zips that defcad had to remove, if I recall.

Since github is in the US a mirror on gittorrent, keybase, and/or just
non-US git hosters may be a good idea.

grendal87, 2020-01-27:
I think I only have the files here. I was thinking pirate bay on its many mirrors might be a good idea in the event of a total ban. 

Pirate bay is like a hydra. Cut off the head and 2 more grows back. 

I was also thinking of getting the weapons guild backed up along with files,books, videos and anything else from the people there. I do not have a big enough hard drive yet for that though.

maduce, 2020-03-14:
The files here replace previous repos including those on the torrent. updates coming soon.

ronnielewis46, 2020-03-31:
Where is the file location? I cannot find it

jungletek, 2020-04-04:
> Where is the file location? I cannot find it

Go to the main page for this repo (https://github.com/maduce/fosscad-repo/) and press the 'Clone or Download' button, then press 'Download ZIP' in the pop-up that opens.

grendal87, 2020-04-04:
If they are on a mobile device they need to set it to request desktop site in order for it to load properly. Without it you cannot download the zip.