maduce/fosscad-repo #45: Guide on how to set everything up

lostforestguy, 2022-02-22:
hi everyone
i am new to this idea so i have many questions

what 3d printer should i get to correctly print this bump stock?

what type of AR15 must I have for the bump stock to mount correctly? colt? Danial defense? Will this bump stock work on AR15 without a buffer tube?

How can i learn how to use a 3d printer?

how can i download this bump stock file?

kasper-sh-hess, 2022-02-22:
You ask "_green glowy newbie_" questions, you know? 
So  - i dont want to be rude - but if this is gonna your first print - I do not recommend that. 
3D printing is not THAT easy and simple. Eg. - right material (filament), right print orientation, right print settings, right printer calibration... etc. Also - you ask about printing something that could be illegal in your country - and is illegal in US - so dont think that whoever will help you directly with printing **that** model. "_Hypotetically_" - yea, but not practically, if you understand what i mean. 
But if you can still deal with it - there's only one way... ONE RIGHT WAY. Go to this site:
https://ctrlpew.com/the-complete-getting-started-guide/
And do every lesson / step.  
Thats all from me. GL

lostforestguy, 2022-02-22:
Thank you so much. 
Do you happen to have the download file for bump stock image?

lostforestguy, 2022-02-22:
I think i found it 
https://github.com/maduce/fosscad-repo/find/master
is there anyone here who could help me with this?

lostforestguy, 2022-02-22:
I have a lot of money so is there someone who can help me?

Triple7even, 2022-02-22:
Lol

grendal87, 2022-02-22:
> hi everyone i am new to this idea so i have many questions

Welcome to 3d printing. 

> what 3d printer should i get to correctly print this bump stock?

First unless you wanna spend over 10 years getting pounded in the butt by another man or forced to suck them off drop the bump stock idea. They are illegal and a jailable offence. 

There has been a number of people arrested for them already. No one here is gonna help you break the law. They would then be looking at a conspiracy charge. 

As for the printer depends what you can afford. 

> what type of AR15 must I have for the bump stock to mount correctly? colt? Danial defense? Will this bump stock work on AR15 without a buffer tube?

Nix the bump stock idea. If you wanna bump fire loop your finger through the belt loop on your jeans and do it that way. You cannot get arrested for that....yet...

> 
> How can i learn how to use a 3d printer?

Well if you wanna design your own files. I sugguest taking cad or computer aided design classes. As for the 3d printing youtube has many good channels for learning to 3d print. I tend to use simplify3d software wise. 

> 
> how can i download this bump stock file?

You don't word is even possessing the files for a bump stock and 3d printing will get you a very watchful eye. I got a visit from the ATF based on an untrue bull story I was dealing guns out of my house. Cost me 5,000 for a lawyer cause I had my ender 5 pro sitting on the table in full view of the ATF. Still waiting to get my guns back 2 years later and numerous trips back to court. Just getting my external hard drive and 3d printer was like pulling teeth. 

So nix the idea of doing anything illegal.

lostforestguy, 2022-02-22:
you wish i would surrender and go to prison motherfucker, i would rather be dead then let those cocksuckers kidnap me.
About the files, they are right here
https://github.com/maduce/fosscad-repo/tree/master/Rifles/AR-15_Bumpfire_Stock_v2-Disruptive_Solutions
it appears i can download them.

lostforestguy, 2022-02-22:
i just have to learn how the ins and out of it all which will take time.

Triple7even, 2022-02-22:
Bro just go on, you sound like a hungry ass FED

lostforestguy, 2022-02-22:
if i was a fed why the fuck would i be asking cocksuckers like you for help

lostforestguy, 2022-02-22:
the only person here who has remotely helped was kasper-sh-hess

crushedpp, 2022-02-22:
Hey man,

None of us are going to help you do something illegal. I don’t know
anything about 3D printing but you may want to look into BOFA for a good
CAD/design program. I use it all the time.

tekopsinc, 2022-02-22:
In mississippi the bumpstock ban won't be enforced, it's legal at the state level, he may be from mississippi...  https://thenewamerican.com/south-dakota-allows-constitutional-carry-mississippi-fights-bump-stock-ban/

grendal87, 2022-02-22:
> you wish i would surrender and go to prison motherfucker, i would rather be dead then let those cocksuckers kidnap me. About the files, they are right here https://github.com/maduce/fosscad-repo/tree/master/Rifles/AR-15_Bumpfire_Stock_v2-Disruptive_Solutions it appears i can download them.

Again.....NIX THE BUMPSTOCK IDEA. 

You don't need it to bump fire. I been involved with firearms and 3d printing for 22 years. The bump stock did nothing but give people who cannot understand the physics an option to have fun simulating automatic fire. You can do it with your jeans. You can do it with rubber bands if you really wanted to. The best thing with those options is you won't be arrested for it. 

I am not talking binary triggers which are going the same way as the bump stock. The bump stock also did not work with a piston as you think. 

The FRT did not require full auto bolt carrier, full auto spring, full auto trigger. and other misc. The reason it was recommended to take it to a gunsmith for installation was because of issues in some weapons where the FRT trigger system had clearance issues that required a knowledgeable gunsmith to fix those clearance issues where you the laymen dont destroy the gun entirely.

For example I worked on one where the FRT didnt fit correctly and I had to correct the geometry of the hammer.  I had to remove 3 thousandths of an inch of material while keeping the rest of the geometry. For reference the thickness of a human hair is 6 thousandths of an inch. 1 extra pass would of ruined the hammer in the customers gun. 

Also bump stocks were prone to higher rates of failure. Your taking and increasing the wear on a weapon that wasnt designed for it. You had hammers riding the bolt, firing pin breakages, busted springs and more leading to stoppages. 

You have already crossed a legal line that can get you arrested and some of your statements are alarming in nature and will resort in you being reported. We will not condone or assist you in breaking the law.

lostforestguy, 2022-02-22:
your a total fucking retard, bump stocks worked better on a gas piston rifle, a well designed piston driven rifle will last well, well over 1000 rounds before failing and a DI blowback rifle will fail much sooner, i have shot 1000s of rounds out of both rifles

about the frt 15 triggers, your a fucking retarded bitch, it needs full auto parts to work or it will be a jamming piece of shit

go ahead and report me bitch i dont give a fuck

meunlo, 2022-02-22:
Lol grovels for help. Then calls everyone retarded bc they don't know more
than the one asking for help. If so smart? Why you asking for help...
HILARIOUS. the internet never fails to provide entertainment and your
plight doesn't fall short. CHEERS

meunlo, 2022-02-22:
Also. You're inability to spell you're* correctly, just slaps. Thanks for
this!

lostforestguy, 2022-02-22:
stay mad loser

meunlo, 2022-02-22:
YOU'RE PROJECTION IS ON POINT.  I'M 💀☠️

lostforestguy, 2022-02-22:
bump stocks are better then frt and binary trash
want proof?

lostforestguy, 2022-02-22:
Stephen Paddock

tactiham, 2022-02-22:
This is the most entertaining spam email chain I have ever witnessed.

As for the idiot who started this, please keep the smooth brained glow,
you're hilarious.

Thank you, y'all.

lostforestguy, 2022-02-22:
Stephen Paddock>frt15>binary trigger

lostforestguy, 2022-02-22:
"I am not talking binary triggers which are going the same way as the bump stock" you think the atf is gonna ban them?

grendal87, 2022-02-22:
> "I am not talking binary triggers which are going the same way as the bump stock" you think the atf is gonna ban them?

The ATF has already banned the binary trigger. The ATF has raided people already because biden did the same thing trump did. Signed an executive order and the ATF changed its definition to suit biden. 

The ATF concluded the Rare Breed triggers to be a combination of parts designed and intended to convert a semi-automatic weapon into a machine gun. The ATF’s examination found that the Rare Breed trigger allows a firearm to “shoot, automatically more than one shot, without manual reloading, with a single continuous pull of the trigger.” 

So in the eyes of the people that will ruin your life Bump stocks, and binary triggers are machine guns. Possession of such is illegal. Assisting in manufacturing them is illegal. 

Insulting us is not gonna change the answer from no to yes. 

So either you one of these guys pictured below: 
[image]

Or your a wack job looking to commit a mass shooting and cant figure out how to go about obtaining illegal weapons to do it.

meunlo, 2022-02-22:
🍿🍿🍿🍿🍿

RWBRITT, 2022-02-22:
[image]

grendal87, 2022-02-22:
> [image]

I will lick the spoon if need be. Just a gut feeling something is off with this guy.

RWBRITT, 2022-02-22:
Like he is an agent?

grendal87, 2022-02-22:
> Like he is an agent?

Honestly after the Stephen paddock comments....im thinking more leakage from a would be mass shooter. An agent would of buggered off the moment they met backlash....like the numerous gun walking schemes.