maduce/fosscad-repo #47: Hello, guys, are there assembly instructions? thx

i0S, 2022-12-03:
Let's say the M1911, I don't know how to put up all the components all together.
thanks a lot.

grendal87, 2022-12-03:
There is a magical place.....called the tube of you...shows you many things one is stuff like this. https://youtu.be/4P4XJTKzKEU
Also if your having trouble with the 1911 might not be the hippy for you as the 1911 is super simple. Do not break the law.

i0S, 2022-12-03:
> There is a magical place.....called the tube of you...shows you many things one is stuff like this. https://youtu.be/4P4XJTKzKEU Also if your having trouble with the 1911 might not be the hobby for you as the 1911 is super simple. Do not break the law.

:D
I'm playing world of guns on steam. I know the assembly there. But I found the fosscad version has less components. Doesn't they supply all the parts, according to laws or some rules? thanks again.

grendal87, 2022-12-03:
No they do not supply all the parts. You buy a parts kit that has the parts that don't come with the 3d printing files. You cannot currently at home 3d print every single part.

i0S, 2022-12-04:
> 

I see. If I just want a toy as many detachable components as possible, no need to shoot, could u recommend a model I can print at home? 
thanks a lot.